PubInv/krake #127: HArdware, The PCB designers need a vertical distance of DE9 compatable with the enclosure.

nk25719, 2024-12-29:
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 The PCB designers need a vertical distance of DE9 compatable with the enclosure.
 

**Additional context**
_Add any other context or screenshots about the feature request here._

[image]

[image]

ForrestErickson, 2025-02-03:
Found a part parameter at Digitkey that shows the spacing from the mounting flange to the board lock.

https://www.digikey.com/en/products/detail/amphenol-cs-fci/LD09S13A4GX00LF/4997291

[image]

nk25719, 2025-04-27:
Refs: 

Technical specification: /https://www.amphenol-cs.com/media/wysiwyg/files/drawing/10130913.pdf 

Amphenol doc: /https://cdn.amphenol-cs.com/media/wysiwyg/files/documentation/customerpresentation/d_sub_productpresentation.pdf

Amphenol Product Catalog: /https://cdn.amphenol-cs.com/media/wysiwyg/files/documentation/datasheet/inputoutput/io_dsub_brochure.pdf

[image]

PN of DB9 used in Krake: LD09S13A4GX00LF

[image]

[image]

closed 2025-04-27 by nk25719

nk25719, 2025-04-27:
Ref for the DSUB Punch for the front panel of the enclosure.