PubInv/krake #131: Firmware, Unexpected characters at the end of the alarm message, GPAD API 0.22

nk25719, 2025-01-05:
**Describe the bug**
_A clear and concise description of what the bug is._
Short Summary of the expected behavior: LCD should display message as sent.
Short Summary of the buggy behavior:  LCD had unexcpected characters at end of message.
 [image]
 
**To Reproduce**
Steps to reproduce the behavior:
1. using the MightyIT MQTT terminal pro application

2. Sending a message with CR and LF check boxes selected
[image]

3. See extra characters on the LCD. See screenshot at the "Short Summary of the buggy behavior:  " above.
 

 
**Hardware description**
Model: KRAKE
Serial Number: USA2, USA1, LEB1

 
**Test Equipment**
Model: MightyIT MQTT terminal pro
Serial Number: Phone in LEB

**Additional context**
_Add any other context about the problem here._
Series of screenshots with and without CR and LF 

[image]
>>App interface used

[image]
[image]
[image]
[image]
>>All combinations with and without CR and LF

[image]
>>Characters on LCD

RobertLRead, 2025-01-06:
A fix was attempted in code version 0.25 that seems to work.

nk25719, 2025-01-06:
[image]

closed 2025-01-06 by nk25719

ForrestErickson, 2025-01-07:
## Preproducing the Failure with PMD_Processing_MQTT
Lee modified PMD_Processing_MQTT to version 0.28 Add test 't' sends "a1Test with CR and LF\r\n" for reproducing  Unexpected characters at the end of the alarm message, GPAD API 0.22 #131

## PMD V0.28 Can reproduce the failure.
When testing GPAD_API V0.24 and the modified PMD the bug is reproduced.
[image]