PubInv/krake #142: Hardware, ESP32 Memory Size, What Do We Need

ForrestErickson, 2025-01-23:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
 
We want to build Krake with ESP32 and have to choose between -N4, -N8 and -N16 memory size.
We do not know the size of what our development kits memory, so we do not know what to order from JLCPCB
 
The datasheet for the N4 says it is not approved for new design.

How much memory do we need?

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
We want to know what to use.

ForrestErickson, 2025-01-23:
An Ai found a sketch which returns the memory size.

```
void setup() {
    Serial.begin(115200);
    while (!Serial);

    // Print flash size
    Serial.println("Flash Size:");
    Serial.println(ESP.getFlashChipSize() / (1024 * 1024));
}

void loop() {
    // Do nothing
}
```

Lee compiled and loaded and got:

> 13:04:57.439 -> Flash Size:
> 13:04:57.490 -> 4

The micro controllers Lee has return 4 (Meg)
The micro controllers Nagham has return 4 (Meg)

RobertLRead, 2025-01-23:
I believe 4 megabytes is more than enough for our anticipated usage.