PubInv/krake #144: Doc - PCBWAY Order History - KRAKE REV1 PW ASSY

nk25719, 2025-01-31:
Requested order status for (by sending an email and a message on the website):

-  Product No W732620AS1C9
-  Product No.:T-1C10W732620A

We updated the shopping cart with new gerber files for the PCB Product No W732620AS1C9, and added assembly information (BOM and centroid files.) Assembly for PCB Product No.:T-1C10W732620A

Please advise on the next steps.

[image]

### URL for PCBWAY shopping car: 
>https://member.pcbway.com/Order/CartList/

nk25719, 2025-02-07:
[PCBa_KRAKEKiCad6_250128_1359_BOM(2025-02-06).xls](https://github.com/user-attachments/files/18695293/PCBa_KRAKEKiCad6_250128_1359_BOM.2025-02-06.xls)

PCBWAY sent us a quotation and part notes and we captured the sheet including our reply here.

nk25719, 2025-02-07:
[image]

Placed order and processed the payment.

ForrestErickson, 2025-03-10:
## Questions from PCBWAY
Nagham received email from PCBWAY

_<html>
<body>
<blockquote style="color: rgb(85, 85, 85); font-size: 13px; border-left: 2px solid rgb(221, 221, 221); padding: 0px 0px 0px 16px; margin: 16px 0px; font-family: -apple-system, BlinkMacSystemFont, &quot;Helvetica Neue&quot;, &quot;Segoe UI&quot;, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;, &quot;Segoe UI Symbol&quot;; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><div dir="ltr"><b>From:</b><span> </span>PCBWay Online Services Team &lt;<a href="mailto:service31@pcbway.com" class="mailto-link" target="_blank" style="word-break: break-word; text-decoration: underline; color: rgb(60, 97, 170);">service31@pcbway.com</a>&gt;<br><b>Date:</b><span> </span>10 March 2025 at 04:16:38 GMT+2<br><b>To:</b><span> </span><a href="mailto:naghamkheir25@gmail.com" class="mailto-link" target="_blank" style="word-break: break-word; text-decoration: underline; color: rgb(60, 97, 170);">naghamkheir25@gmail.com</a><br><b>Subject:</b><span> </span><b>PCBWay--Plz check the pics of PCBA order T-1C10W732620A</b><br><br></div></blockquote><blockquote class="collapsed-blockquote" id="collapsed-blockquote-1387" style="color: rgb(85, 85, 85); font-size: 13px; border-left: 2px solid rgb(221, 221, 221); padding: 0px 0px 0px 16px; margin: 16px 0px; font-family: -apple-system, BlinkMacSystemFont, &quot;Helvetica Neue&quot;, &quot;Segoe UI&quot;, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;, &quot;Segoe UI Symbol&quot;; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><div dir="ltr">﻿<div style="font-size: 14px; color: rgb(0, 0, 0); font-family: Verdana;"><span style="font-size: 14px;"><span style="color: rgb(0, 0, 0);"><span style="font-family: Verdana;"><div style="font-size: 14px; color: rgb(0, 0, 0); font-family: Verdana;"><span style="font-size: 14px;"><span style="color: rgb(0, 0, 0);"><span style="font-family: Verdana;"><span style="font-size: 14px;"><span style="color: rgb(0, 0, 0);"><span style="font-family: Verdana;"><span style="font-size: 14px;"><span style="color: rgb(0, 0, 0);"><span style="font-family: Verdana;"><span style="font-size: 14px;"><span style="color: rgb(0, 0, 0);"><span style="font-family: Verdana;"><span style="font-size: 14px;"><span style="color: rgb(0, 0, 0);"><span style="font-family: Verdana;"><span style="font-size: 14px;"><span style="color: rgb(0, 0, 0);"><span style="font-family: Verdana;"><span style="font-size: 14px;"><span style="color: rgb(0, 0, 0);"><span style="font-family: Verdana;"><span style="font-size: 14px;"><span style="color: rgb(0, 0, 0);"><span style="font-family: Verdana;"><span style="font-size: 14px;"><span style="color: rgb(0, 0, 0);"><span style="font-family: Verdana;"><span style="font-size: 14px;"><span style="color: rgb(0, 0, 0);"><span style="font-family: Verdana;"><span style="font-size: 14px;"><span style="color: rgb(0, 0, 0);"><span style="font-family: Verdana;"><span style="font-size: 14px;"><span style="color: rgb(0, 0, 0);"><span style="font-family: Verdana;"><span style="font-size: 14px;"><span style="color: rgb(0, 0, 0);"><span style="font-family: Verdana;"><span style="font-size: 14px;"><span style="color: rgb(0, 0, 0);"><span style="font-family: Verdana;"><span style="font-size: 14px;"><span style="color: rgb(0, 0, 0);"><span style="font-family: Verdana;"><span style="font-size: 14px;"><span style="color: rgb(0, 0, 0);"><span style="font-family: Verdana;"></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span></span>Hi,<br><br>Good day.<br>Attached plz find the pics of finished PCBA and pay attention to the pics' name.<br><br><span style="font-family: Verdana; font-size: 14px; white-space: normal; color: rgb(0, 0, 0);">As it is just for you to confirm the orientation or some other details, we didn't solder all the pins in case there's any problem. </span><br><br>Your prompt and earliest reply would be greatly appreciated.<br><div><br></div></span></div></span></span></span></div><span style="color: rgb(204, 204, 204); font-size: 12px;"> 2025-03-10 </span><br><span style="color: rgb(204, 204, 204); font-size: 12px;">------------------------------<wbr>-----</span><br><br>
Best Regards
--
 
Judith Yang
Sales account manager
E: service31@pcbway.com
Teams: service31@pcbway.com
Tel: 0571-8709-3808  GMT+8: 9:00 - 18:00, Mon.- Sat.(12a.m.)

</div></blockquote>
</body>
</html>From: PCBWay Online Services Team <[service31@pcbway.com](mailto:service31@pcbway.com)>
Date: 10 March 2025 at 04:16:38 GMT+2
[... cut ...]
Lee produced the following feedback for Nagham

> Nagham,
>  
> The orientation of D101 is correct.
> The orientation of D104 is correct.
> The orientation of D106 is correct.
> The orientation of D501 is correct.
> The orientation of U301 is correct.
>  The orientation of U501 is correct.
> The orientation of U701 is correct.
>  
> The placement of S101 is correct
> The placement of S102 is correct
> The placement of S601 is correct
> The placement of RV301 is correct
>  
> I think this might answer all of their concerns.
> Lee

nk25719, 2025-03-12:
[image]

nk25719, 2025-04-04:
Order received. Closing Issue.

closed 2025-04-04 by nk25719