PubInv/krake #148: Doc - JLCPCB Order History - KRAKE REV1 PW ASSY, Lee

ForrestErickson, 2025-02-03:
20250203 1410

Placing order at JLCPCB:

Added the Gerber files:

Gerbers_250128_1359.zip
Added BOM and CPL files:

PCBa_KRAKEKiCad6_250128_1359_BOM_JLC.csv
PCBa_KRAKEKiCad6_250128_1359_CPL_JLC.csv

Choose parameters including Confirm Production File
[image]

[image]

## Set assembly options.
Choose Confirm Parts Placement and get dialog and choose NOT to confirm automatically
[image]

Choose Depanel boards & edge rail before delivery

[image]

Pressed next.

## Board Render
[image]  
Pressed next.

## Upload BOM/PL and CTL Files
[image]
Press: Process BOM and CPL

Get the Error dialog for parts not populated.
[image]  
Pressed Continue.

For J???replace with C492401
For J601 replace with C703015
For all 330R replace with  RS-03K331JT
For all 4K7 replace with C23162

This cleared all shortfalls.   
[image]  
So press next

[image]
The render finalized.  

[image]

ForrestErickson, 2025-02-05:
## Order Emails from JLCPCB
2/4/2025 5:37 PM  
[image]

Second email
2/4/2025 7:46 PM   
[image]

## Visit link and approve
[image]

Lee approved the production but sent two emails for non critical clarification.
### First Email  re JLCJLCJLCJCL
[image]
### Second Email re 2D code
[image]

ForrestErickson, 2025-02-07:
Another request to confirm proeuction

[image]

[image]

The dialog to enter parts placement review
[image]

The dialog for parts placement
[image]

I gave instructions for rotation of three diodes.  
[image]

ForrestErickson, 2025-02-08:
Email on 

```
On 02/08/2025 9:59 AM EST JLCPCB <noreply@jlcpcb.com> wrote:
Dear ForrestErickson,

The parts placement of your order  SMT025020361767 is completed. The order is ready for manufacturing now. Please check and confirm it through your [order history](https://mandrillapp.com/track/click/31015052/jlcpcb.com?p=eyJzIjoiUm9xaFdmUmRsQ3ZSR3A0dDU4Z0UtcHBqMTZRIiwidiI6MSwicCI6IntcInVcIjozMTAxNTA1MixcInZcIjoxLFwidXJsXCI6XCJodHRwczpcXFwvXFxcL2psY3BjYi5jb21cXFwvdXNlci1jZW50ZXJcXFwvb3JkZXJzXCIsXCJpZFwiOlwiZTU3MjY4N2E0ZGQ1NDY0ZWI2NjYxYjMxOTAxY2Q1ZWZcIixcInVybF9pZHNcIjpbXCI0YmE5OGExMDZjMGZhMjQ3NzBmMjE0Y2Y3NDQ2Y2EzOTQ0MWIyMmM3XCJdfSJ9).

```

## My review response
Hello and thank you for rotating the parts. 
However now I am feeling confused and realized I should have asked how the plus and minus symbols on the 3D view are showing anode and cathode.  I may have been wrong.

To clearly state my requirement 
D101 Cathode up.
D103 Cathode right.
D104 Cathode up.

Sorry to have to revisit this but I confused my self.

ForrestErickson, 2025-02-12:
On 20250212 Lee sent email to JLCPCB

> Hello Support,
>  
> Regarding, SMT025020361767-2999181A
>  
> I am not sure of the status of this order. 
> Is your system waiting for me to confirm something or not?
> Here is a screen shot about which I am confused.
> 
>  
> Please advise.
>  
> Thanks,
> Forrest Lee Erickson
>  
> CC: Public Invention Team

ForrestErickson, 2025-02-12:
On 20250212 Lee Placed an Amazon order resulting in two Invoices

[Amazon.com - Order 112-3993719-9023403.pdf](https://github.com/user-attachments/files/18771701/Amazon.com.-.Order.112-3993719-9023403.pdf)

[Amazon.com - Order 112-9690873-4379401.pdf](https://github.com/user-attachments/files/18771703/Amazon.com.-.Order.112-9690873-4379401.pdf)

ForrestErickson, 2025-02-13:
Email from JLCPCB

> On 02/13/2025 3:21 AM EST sinclairzhu@jlcpcb.com wrote:
> Hi Forrest Lee Erickson,
>  Thanks for your reply.
>  Our engineer said the plus symbol means anode and the minus symbol means cathode, he had rotated  D101, D103, D104 as you required, if it's OK, you can confirm the parts placement in your order history. Thank you.
> 

[image]

Lee visited the link and found the confirm button and pressed it.
Sent email back

> On 02/13/2025 12:24 PM EST FORREST ERICKSON <forresterickson@comcast.net> wrote:
> Hello Sinclair,
> Thanks for the clarification and the image.
> I have approved the placement.
> Have a good day!
> Forrest Lee Erickson
> CC: Public Invention Team
>

ForrestErickson, 2025-02-14:
An email from JLCPCB

> On 02/13/2025 8:05 PM EST sinclairzhu@jlcpcb.com wrote:
> Hi Forrest Lee Erickson,
>  Thanks for your reply.
>  OK, this order is ready for production now, thank you.
>  Best regards,
>  Sinclair
> 

CC: @nk25719

ForrestErickson, 2025-04-04:
<html>
<body>
<div>On 02/17/2025 10:04 PM EST JLCPCB &lt;noreply@jlcpcb.com&gt; wrote:</div>
<div> </div>
<div> </div>
[image]
<p>Dear ForrestErickson,</p>
<p> </p>
<p>Thank you for your order from JLCPCB. It’s our pleasure serving you. We are pleased to inform you that the items listed below have been packaged and are waiting for pick up by DHL EXPRESS. Tracking Number: 1923805682</p>
<p> </p>
<p>Kindly note that it may take 1-2 days before carriers will display tracking information.</p>
<p> </p>
<p>Please track the shipment in your <a href="https://mandrillapp.com/track/click/31015052/jlcpcb.com?p=eyJzIjoiY0FyRW1Zb2tFby1iS056dld4Q2tQU3YxN3F3IiwidiI6MSwicCI6IntcInVcIjozMTAxNTA1MixcInZcIjoxLFwidXJsXCI6XCJodHRwczpcXFwvXFxcL2psY3BjYi5jb21cXFwvdXNlci1jZW50ZXJcXFwvb3JkZXJzXCIsXCJpZFwiOlwiY2NmZDRhYjg0ZWMwNDllMzgzMjkwZmYxOWE1NGMwYzJcIixcInVybF9pZHNcIjpbXCI0YmE5OGExMDZjMGZhMjQ3NzBmMjE0Y2Y3NDQ2Y2EzOTQ0MWIyMmM3XCJdfSJ9" target="_blank" rel="noopener">order history</a>.</p>
<p> </p>
<p><span style="color: #606266;">Items in this shipment：【W202502050208228】</span></p>
<p> </p>

  | Gerbers_250128_1359_Y20 5 pcs
-- | --
  | 5 pcs
  | View order details>>

</body>
</html>
Order received in March. On 02/17/2025 10:04 PM EST JLCPCB <noreply@jlcpcb.com> wrote:
Dear ForrestErickson,
Thank you for your order from JLCPCB. It’s our pleasure serving you. We are pleased to inform you that the items listed below have been packaged and are waiting for pick up by DHL EXPRESS. Tracking Number: 1923805682
Kindly note that it may take 1-2 days before carriers will display tracking information.
Please track the shipment in your [order history](https://mandrillapp.com/track/click/31015052/jlcpcb.com?p=eyJzIjoiY0FyRW1Zb2tFby1iS056dld4Q2tQU3YxN3F3IiwidiI6MSwicCI6IntcInVcIjozMTAxNTA1MixcInZcIjoxLFwidXJsXCI6XCJodHRwczpcXFwvXFxcL2psY3BjYi5jb21cXFwvdXNlci1jZW50ZXJcXFwvb3JkZXJzXCIsXCJpZFwiOlwiY2NmZDRhYjg0ZWMwNDllMzgzMjkwZmYxOWE1NGMwYzJcIixcInVybF9pZHNcIjpbXCI0YmE5OGExMDZjMGZhMjQ3NzBmMjE0Y2Y3NDQ2Y2EzOTQ0MWIyMmM3XCJdfSJ9).
Items in this shipment：【W202502050208228】
Gerbers_250128_1359_Y20
5 pcs
5 pcs

	[View order details>>](https://mandrillapp.com/track/click/31015052/jlcpcb.com?p=eyJzIjoiaGtkQkNqZ2ZpMG9uX042QUxRRWJYYzRGQzlvIiwidiI6MSwicCI6IntcInVcIjozMTAxNTA1MixcInZcIjoxLFwidXJsXCI6XCJodHRwczpcXFwvXFxcL2psY3BjYi5jb21cXFwvdXNlci1jZW50ZXJcXFwvb3JkZXJzXFxcL1wiLFwiaWRcIjpcImNjZmQ0YWI4NGVjMDQ5ZTM4MzI5MGZmMTlhNTRjMGMyXCIsXCJ1cmxfaWRzXCI6W1wiNGJhOThhMTA2YzBmYTI0NzcwZjIxNGNmNzQ0NmNhMzk0NDFiMjJjN1wiXX0ifQ)

nk25719, 2025-04-27:
Check issue #168 to see the unpacking for the received order. 

Closing. @ForrestErickson

closed 2025-04-27 by nk25719