PubInv/krake #149: Hardware, Schematic and BOM (PCBa_KRAKEKiCad6_250128_1359_BOM) Errors

ForrestErickson, 2025-02-06:
Error in file: PCBa_KRAKEKiCad6_250128_1359_BOM

D106 is a 0603 LED but has incorrect manufacturer part number calling out an 0805.
The JLCPCB part number correct for an 0603.

Manufacturing Part number with error
[image]

The rest of the fields should be double checked too.

nk25719, 2025-02-06:
[image]

Incorrect MPN, the correct MPN is: RMC060310KFN

We approved with PCBWAY to use: FRC0603F1002TS for manufacturing PCBa Rev1

nk25719, 2025-02-07:
This Datasheet info for the Enclosure is obviously wrong,

[image]

nk25719, 2025-02-13:
After fixing the above mentioned errors we need to regenate the BOM and gerbers for the future orders.

nk25719, 2025-02-27:
[CustomerReply_20250227_T-1C10W732620A.xlsx](https://github.com/user-attachments/files/19015147/CustomerReply_20250227_T-1C10W732620A.xlsx)

Additional BOM errors corrected for PCBWAY, 
TP101 
and J601 

(We need to update the BOM sheet before sending out another order.)

nk25719, 2025-04-27:
Additional corrections and data routing for the MPN and manufacturer for the DB9 used: 

[image]

closed 2025-04-27 by nk25719