PubInv/krake #151: Hardware, Designing a Boss for a 3mm Screw for 3D Printed Enclosures (PLA and more)

nk25719, 2025-02-08:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
There was no thread engagement on the screws which mounted the PCB into the GPAD's enclosure. 
 
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 A 3mm plastide screw mount the printed circuit board to the enclosure.
 
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
 Use threaded inserts.

samoyin, 2025-02-08:
I have already created and rendered a 3mf and stl file for the Boss. 
See Enclosure folder.

ForrestErickson, 2025-02-08:
## Boss Design Evaluation Steps

- [ ] Order 3mm x 8mm Plastite screw
- [ ] Print file in PLA+
- [ ] Test to destruction
- [ ] Print in ABS+
- [ ] Test to destruction
- [ ] Print in PTEG
- [ ] Test to destruction

closed 2025-11-21 by nk25719