PubInv/krake #154: Software, IP Address Not Reported, Version 0.25

ForrestErickson, 2025-02-13:
**Describe the bug**
_A clear and concise description of what the bug is._
Short Summary of the expected behavior:  See the IP address of the Krake once connected to the WiFi network

Short Summary of the buggy behavior:  Not reported on serial port.
 
 Serial Monitor capture of Boot and WiFi Connection

```
13:00:05.898 -> PubInv GPAD: enter command in form CDa (C is a char, D is a digit)
13:00:06.006 -> ===================================
13:00:06.006 -> KRAKE_
13:00:06.006 -> GPAD_API 0.25 
13:00:06.006 -> Builtin ESP32 MAC Address: 3C61053DF08C
13:00:06.006 -> Alarm Topic: 3C61053DF08C_ALM
13:00:06.006 -> Broker: public.cloud.shiftr.io
13:00:06.006 -> Compiled at: Jan 23 2025 13:16:54
13:00:06.006 -> GNU Affero General Public License, version 3 
13:00:06.006 -> ===================================
13:00:06.052 -> 
13:00:07.203 -> Done With Setup!
13:00:12.939 -> Publish RSSI: 0.00
13:00:22.977 -> 
13:00:22.977 -> Connecting to WiFi: VRX
13:00:22.977 -> Failed to connect WiFi.
13:00:22.977 -> Attempting MQTT connection...failed, rc=-2Attempting MQTT connection...failed, rc=-2Attempting MQTT connection...failed, rc=-2failed to reconnect!
13:00:25.992 -> Publish RSSI: -55.00
13:00:32.957 -> Publish RSSI: -56.00

```

ForrestErickson, 2025-05-22:
For some time the IP address is reported on the serial monitor.
It should be displayed, reported on the LCD scree for the end users access.

nk25719, 2025-05-31:
IP address is reported on the LCD screen. 

Closing.

closed 2025-05-31 by nk25719