PubInv/krake #155: Doc, Factory Test and Assembly Instructions.

ForrestErickson, 2025-02-14:
# Test and Assembly Documentation

 Write a test procedure where there is testing before power is applied, with power applied, with a factory test firmware and finally loading final firmware.

Write so that it will be easy to incorporate into a HardwareX article.

CC: @RobertLRead

ForrestErickson, 2025-02-14:
Robert,

I hope you can joint us at 12:00 ET today in Discord.
I want to talk about writing a factory test and assembly document for Krake and realized that I should be considering how to do so for lowest effort including a HardwareX article.

Nagham,
Please see the GPAD markdown document: Manufacturing and Unit Test Documentation, PCB Version 2.0, 20230228 
at:  https://github.com/PubInv/general-purpose-alarm-device/blob/main/Hardware/Documentation/ManufacturingUnitTestTroubleshootingRev2.md

For GPAD, there is also a HardwareX document at:
https://github.com/PubInv/general-purpose-alarm-device/tree/main/doc/
See: HardwareXRevised.docx

Even thought Robert did the bulk of the work, there was considerable struggle against the "helpful" editing tools of Word and OpenOffice to get documentation from MarkDown into the HardwareX article that I do not every want to experience again 

FYI,
I found an online tool which converts ODT to Markdown here:
https://www.vertopal.com/en/download#379e90594779457dbc2e0ab806de143a

I tried it just now. It converted the HardwareXRevised.docx to with some errors. The errors appear to be related to formatting of size of images.

Lee

nk25719, 2025-02-18:
Create file testing for: 
PCBs 
Firmware 
Enclosure

ForrestErickson, 2026-02-12:
# Document Location 
The Rev2 PWA [Krake_Test_and_Assembly_Procedure.md](https://github.com/PubInv/krake/blob/155-doc-factory-test-and-assembly-instructions/PWA_REV2/Krake_Test_and_Assembly_Procedure.md)

courtneyludick, 2026-02-12:
# Document Location with 'blob'
https://github.com/PubInv/krake/blob/Enhancement/155-doc-factory-test-and-assembly-instructions/PWA_REV2/Krake_Test_and_Assembly_Procedure.md

nk25719, 2026-02-12:
https://github.com/PubInv/krake/blob/main/PWA_REV2/Krake_Test_and_Assembly_Procedure.md

courtneyludick, 2026-02-19:
https://github.com/user-attachments/assets/e9746968-028b-4ea7-9159-5aacf3a627d9