PubInv/krake #156: Test Fit, 3mm Machine Screw in to Boss

ForrestErickson, 2025-02-15:
Tape the boss to a table top.
[image]

Try with 2.5mm screw
[image]

ForrestErickson, 2025-02-15:
Lee testing with a plastite like screw taken from a multimeter. The major diameter is 2.63mm  and the minor of 2.30mm.
[image]
I was able to get some thread engagement but

nk25719, 2025-02-15:
[image]

> 3D print of the Boss

[image]

> Trying to fit in a 3 mm Screw

[image]

> Trying to fit in a 3.5 mm Screw

[image]

> Outer diameter of the screw

[image]

> hole diameter 

[image]

> counter bore diameter

ForrestErickson, 2025-02-15:
## Testing Rev2 Boss
File name: BOSS_ONLY_FILE_REV2.3mf
Lee Printed out of PLA. 
Boss OD is 4.79mm
Boss length is 12.76mm
Hole ID is 2.54 mm
Counter Bore ID is 3.09 mm

### Fit to Screw
[image]
A 3 mm machine screw fit into the counter bore.

Turning the machine screw it began to cut threads with some considerable force. 
[image]
 Turned in about three revolutions.

closed 2025-11-21 by nk25719