PubInv/krake #161: Hardware, PCBWAY notes: "Component is bigger then drill, unable to plug in."

nk25719, 2025-02-27:
**Describe the Failure / Bug**
_A clear and concise description of what the failure / bug is. Describe the problem NOT the fix._

 PCBWAY sent us the following information: "Component is bigger then drill, unable to plug in."
This is the component called out in the bill of material for TP101 : RH-5121	TestPoint:TestPoint_Loop_D2.54mm_Drill1.5mm_Beaded

They included this following photograph:
[image]
TP101 is in the photograph
**Hardware description**:
Model: KRAKE Rev1
Serial Number: NA

nk25719, 2025-02-27:
Another instance, Boards at USA, Built by JLCPCB:

[image]

The two parts side by side
Control assembly 1.2 showing the RH-5121 (green) and the RH-5000 (red).
Therefore both sizes are used on the Control assembly 1.2 and was built by JLCPCB.

nk25719, 2025-02-27:
[image]

Datasheets of the two parts side by side, 
on the left the datasheet for the part called out and PCBWAY identified as a problem.
and to the right the smaller part used successfully on HOMEWORK2 assembly by PCBWAY.

nk25719, 2025-02-27:
[CustomerReply_20250227_T-1C10W732620A.xlsx](https://github.com/user-attachments/files/19015105/CustomerReply_20250227_T-1C10W732620A.xlsx)

Resolved the issue by replying to the sheet sent from PCBWAY.

nk25719, 2025-04-27:
The issue was reported only by one of the companies we manufactured the boards with, JLCPCB. The new part numbers for the components in question have been captured in the Excel sheet mentioned in the previous comment. 

Closing this issue as complete.

ForrestErickson, 2025-07-14:
## We Need To Check
Are we calling out the correct test point part number for ALL test points.  The smaller size type of test points.
Is the hole in the PCB appropriate for the smaller size test points.

nk25719, 2025-07-22:
This part should work: LCSC Part # C5243682,  XKB Connection X1311WV-02J-C40D24 , https://www.lcsc.com/product-detail/Pin-Headers_XKB-Connection-X1311WV-02J-C40D24_C5243682.html

nk25719, 2025-07-31:
[image]

ForrestErickson, 2025-08-02:
## JLCPCB Stock Status 202250802
We should be designing with parts commonly stocked.

[image]

ForrestErickson, 2025-08-02:
From the datasheet at: https://www.keyelco.com/userAssets/file/M65p56.pdf
For the Keystone test points with 1.25mm loop, the hole diameter should be 1mm.
[image]

Therefor the KiCad foot print should be the one named: **TestPoint_Loop_D1.80mm_Drill1.0mm_Beaded**

[image]

closed 2025-08-02 by nk25719