PubInv/krake #162: Hardware, PMD, DF Player is not initializing

nk25719, 2025-03-03:
**Describe the Failure / Bug**
The DF Player should initialize and display the menu, however the serial communication is not initializing therefor the module is noy responding if the SD card is found.
 
 
**Device Under Test, aka DUT**
Steps to reproduce the behavior:
1. Model Number: 20240421
2. Serial Number(s): LEB1
3. Location where failure / bug occurred: lebanon
4. Hardware configuration: connect the dfplayer to the board and the speaker to the df player and insert the sd card in the df player.
5. Extent of Failure
** Is failure intermittent? no
** Is failure fixed by power cycling? no
** Is failure permanent? yes
 
 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to PMD_BOOTUP 
2. Click on UPLOAD, THEN Serial Monitor 

 
 
**Screenshots**
_If applicable, add screenshots to help explain your problem._
 
 
[image]

## Hypothesis

This is due to the insufficient voltage provided with 3.3V instead of 5V.

nk25719, 2025-03-04:
Compiled and uploaded another code with a HardwareSerial library and the DF Player responded.

ForrestErickson, 2025-04-25:
Long since fixed.
Closing.

closed 2025-04-25 by ForrestErickson