PubInv/krake #163: Software, Add favicon to the Krake the embedded webpages

nk25719, 2025-03-20:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
The embedded server example from random nerd tutorials does not have a correct PUBINV favicon
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 We need to find a favicon and add to the projects

RobertLRead, 2025-03-21:
Github doesn't allow me to add a .ico file. I have emailed it to you. Please add it to the Krake repo for future reference.

ForrestErickson, 2025-03-22:
There are a limited types of files Github allows to be added. For file types not allowed (or supported or what ever) zip it (them) up.
Like this: 
[PI_favicon.zip](https://github.com/user-attachments/files/19401953/PI_favicon.zip)