PubInv/krake #168: Doc - KRAKE REV1 PW ASSY Finishing and Testing in USA, Lee

ForrestErickson, 2025-04-04:
# Summary
Finishing assembly of Krake in USA using the assemblies from JLCPCB. Ref #148 

## Unpacking

## Assembly Finish

## Assembly Test

ForrestErickson, 2025-04-07:
## Electrical Test

### Unpowered Test 
Tests to make sure no danger to test  equipment power supply
Test with multi meter. In USA make ASTRO /\I DM6000AR

| **Test/DUT**   | **Resistance  TP2-TP1 (Vin)** | **Resistance  TP3-TP1 (+5V)** | **Resistance  TP201-TP1 (VBus)** | **Resistance  J103 Pin 2-TP1 (ControllerVcc)** |
|----------------|-------------------------------|-------------------------------|----------------------------------|------------------------------------------------|
| **Pass Fail**  | Not short,  Capacitive        | Not short,  Capacitive        | Not short,  Capacitive           | Not short,  Capacitive                         |
| **SN: US0001** | >10Meg                        | 1.1K Typical                  | 304 Ohm                          | >10Meg                                         |
|                |                               |                               |                                  |                                                |
|                |                               |                               |                                  |                                                |
|                |                               |                               |                                  |                                                |
|                |                               |                               |                                  |                                                |
|                |                               |                               |                                  |                                                |
|                |                               |                               |                                  |                                                |
|                |                               |                               |                                  |                                                |

### Powered Tests
Apply power with current limited voltage supply.
Minimum input voltage 8V, Maximum 12V.
Current limit TBD Amps.
Apply power at J101.
Measure current and stop test if dangerously high (> 200 mA???)

| **Test/DUT** | Current J101 | TP102 Vin | TP103 +5V | +5esp32 into U103 | +3.3V Tab U103 | TP202 V3 on U501 | D103 Cathode 3v3Controller | C505+ 5VDFP |
|----------------|--------------|-----------|-----------|-------------------|----------------|------------------|----------------------------|-------------|
| **Pass Fail**  | < TMD mA     |           |           |                   |                |                  |                            |             |
| **SN: US0001** | 147mA        | 8.33      | 5.046     | 4.929             | 3.318          | 3.314            | 3.208                      | 5.031       |
|                |              |           |           |                   |                |                  |                            |             |
|                |              |           |           |                   |                |                  |                            |             |
|                |              |           |           |                   |                |                  |                            |             |
|                |              |           |           |                   |                |                  |                            |             |
|                |              |           |           |                   |                |                  |                            |             |
|                |              |           |           |                   |                |                  |                            |             |
|                |              |           |           |                   |                |                  |                            |             |

ForrestErickson, 2025-04-11:
# Functional Test

## With Factory Test Firmware
Load firmware file: TBD 
Test TBD

## With Final Test Firmware
Load firmware: GPAD_API 
Capture information by serial number
Firmware version, MAC, 

| **Serial#** | **MAC** | **Firmware Version** |   | **Note** |
|:-----------:|:-------:|:--------------------:|:-:|:--------:|
|             |         |                      |   |          |
|             |         |                      |   |          |
|             |         |                      |   |          |
|             |         |                      |   |          |
|             |         |                      |   |          |
|             |         |                      |   |          |
|             |         |                      |   |          |
|             |         |                      |   |          |
|             |         |                      |   |          |
|             |         |                      |   |          |
|             |         |                      |   |          |
|             |         |                      |   |          |
|             |         |                      |   |          |
|             |         |                      |   |          |

ForrestErickson, 2025-04-27:
Load Micro SD card with files from:  https://drive.google.com/drive/u/1/folders/1uwaVFpFQ0uJbVLxVXngsslgTf2tzOCR6
See Issue: DOC, Procedure for Loading MP3 Files in SD Card #188

Connect the DUT to a test speaker.

Load firmware: DFPlayerTD5580ATest into DUT 
This firmware can be loaded by USB-serial  interface or by OTA and is OTA ready.

Reset the DUT and confirm audio files play. 
Measure peak audio level with ??? SPL meter.

Confirm peak audio level of ??? +/- ?? db .

|   **Serial#**  | **Peak SPL** |   |   | **Note** |
|:--------------:|:------------:|:-:|:-:|:--------:|
| **SN: US0001** |              |   |   |          |
| **SN: US0002** |              |   |   |          |
| **SN: US0003** |              |   |   |          |
| **SN: US0004** |              |   |   |          |
| **SN: US0005** |              |   |   |          |
| **SN: LB0001** |              |   |   |          |
| **SN: LB0002** |              |   |   |          |
| **SN: LB0003** |              |   |   |          |
| **SN: LB0004** |              |   |   |          |
| **SN: LB0005** |              |   |   |          |
|                |              |   |   |          |
|                |              |   |   |          |
|                |              |   |   |          |
|                |              |   |   |          |

nk25719, 2025-10-01:
Closing as complete.

closed 2025-10-01 by nk25719