PubInv/krake #172: Info, Log Book, Capturing the default Serial Message on A Unprogrammed ESP32

nk25719, 2025-04-10:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
 Capturing the default Serial Message on A Unprogrammed ESP32
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 
 
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
 
 
**Additional context**
_Add any other context or screenshots about the feature request here._

13:51:46.880 -> ets Jul 29 2019 12:21:46
13:51:46.880 -> 
13:51:46.880 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:51:46.880 -> configsip: 0, SPIWP:0xee
13:51:46.938 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:51:46.938 -> mode:DIO, clock div:2
13:51:46.938 -> load:0x3fff0018,len:4
13:51:46.938 -> load:0x3fff001c,len:6932
13:51:46.938 -> load:0x40078000,len:13712
13:51:46.938 -> load:0x40080400,len:4260
13:51:46.938 -> entry 0x400806e4
13:51:46.938 -> [0;32mI (71) boot: Chip Revision: 3[0m
13:51:46.938 -> [0;32mI (72) boot_comm: chip revision: 3, min. bootloader chip revision: 0[0m
13:51:46.985 -> [0;32mI (40) boot: ESP-IDF v4.0.1-193-ge7ac221 2nd stage bootloader[0m
13:51:46.985 -> [0;32mI (40) boot: compile time 02:47:00[0m
13:51:46.985 -> [0;32mI (40) boot: Enabling RNG early entropy source...[0m
13:51:46.985 -> [0;32mI (46) boot: SPI Speed      : 40MHz[0m
13:51:46.985 -> [0;32mI (50) boot: SPI Mode       : DIO[0m
13:51:46.985 -> [0;32mI (54) boot: SPI Flash Size : 4MB[0m
13:51:46.985 -> [0;32mI (58) boot: Partition Table:[0m
13:51:46.994 -> [0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
13:51:46.994 -> [0;32mI (69) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
13:51:46.994 -> [0;32mI (76) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
13:51:46.994 -> [0;32mI (84) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
13:51:47.009 -> [0;32mI (91) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
13:51:47.009 -> [0;32mI (99) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
13:51:47.009 -> [0;32mI (106) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
13:51:47.009 -> [0;32mI (114) boot: End of partition table[0m
13:51:47.009 -> [0;32mI (118) boot_comm: chip revision: 3, min. application chip revision: 0[0m
13:51:47.043 -> [0;32mI (125) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x29230 (168496) map[0m
13:51:47.087 -> [0;32mI (194) esp_image: segment 1: paddr=0x00129258 vaddr=0x3ffbdb60 size=0x03974 ( 14708) load[0m
13:51:47.108 -> [0;32mI (201) esp_image: segment 2: paddr=0x0012cbd4 vaddr=0x40080000 size=0x00400 (  1024) load[0m
13:51:47.108 -> [0;32mI (202) esp_image: segment 3: paddr=0x0012cfdc vaddr=0x40080400 size=0x03034 ( 12340) load[0m
13:51:47.108 -> [0;32mI (215) esp_image: segment 4: paddr=0x00130018 vaddr=0x400d0018 size=0x1068cc (1075404) map[0m
13:51:47.498 -> [0;32mI (604) esp_image: segment 5: paddr=0x002368ec vaddr=0x40083434 size=0x195c8 (103880) load[0m
13:51:47.530 -> [0;32mI (647) esp_image: segment 6: paddr=0x0024febc vaddr=0x400c0000 size=0x00064 (   100) load[0m

ForrestErickson, 2025-04-10:
**US0002**

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6932
load:0x40078000,len:13712
load:0x40080400,len:4260
entry 0x400806e4
[0;32mI (71) boot: Chip Revision: 3[0m
[0;32mI (72) boot_comm: chip revision: 3, min. bootloader chip revision: 0[0m
[0;32mI (40) boot: ESP-IDF v4.0.1-193-ge7ac221 2nd stage bootloader[0m
[0;32mI (40) boot: compile time 02:47:00[0m
[0;32mI (40) boot: Enabling RNG early entropy source...[0m
[0;32mI (46) boot: SPI Speed      : 40MHz[0m
[0;32mI (50) boot: SPI Mode       : DIO[0m
[0;32mI (54) boot: SPI Flash Size : 4MB[0m
[0;32mI (58) boot: Partition Table:[0m
[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (69) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (76) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (84) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32mI (91) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
[0;32mI (99) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
[0;32mI (106) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
[0;32mI (114) boot: End of partition table[0m
[0;32mI (118) boot_comm: chip revision: 3, min. application chip revision: 0[0m

```

ForrestErickson, 2025-04-10:
**US0003**
Serial Monitor of new never programed part

```
19:26:19.296 -> ets Jul 29 2019 12:21:46
19:26:19.296 -> 
19:26:19.296 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
19:26:19.296 -> configsip: 0, SPIWP:0xee
19:26:19.296 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:26:19.296 -> mode:DIO, clock div:2
19:26:19.296 -> load:0x3fff0018,len:4
19:26:19.296 -> load:0x3fff001c,len:6932
19:26:19.296 -> load:0x40078000,len:13712
19:26:19.296 -> load:0x40080400,len:4260
19:26:19.296 -> entry 0x400806e4
19:26:19.296 -> [0;32mI (71) boot: Chip Revision: 3[0m
19:26:19.296 -> [0;32mI (72) boot_comm: chip revision: 3, min. bootloader chip revision: 0[0m
19:26:19.296 -> [0;32mI (40) boot: ESP-IDF v4.0.1-193-ge7ac221 2nd stage bootloader[0m
19:26:19.296 -> [0;32mI (40) boot: compile time 02:47:00[0m
19:26:19.296 -> [0;32mI (40) boot: Enabling RNG early entropy source...[0m
19:26:19.296 -> [0;32mI (46) boot: SPI Speed      : 40MHz[0m
19:26:19.296 -> [0;32mI (50) boot: SPI Mode       : DIO[0m
19:26:19.296 -> [0;32mI (54) boot: SPI Flash Size : 4MB[0m
19:26:19.296 -> [0;32mI (58) boot: Partition Table:[0m
19:26:19.296 -> [0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
19:26:19.296 -> [0;32mI (69) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
19:26:19.296 -> [0;32mI (76) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
19:26:19.296 -> [0;32mI (84) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
19:26:19.296 -> [0;32mI (91) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
19:26:19.296 -> [0;32mI (99) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
19:26:19.296 -> [0;32mI (106) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
19:26:19.296 -> [0;32mI (114) boot: End of partition table[0m
19:26:19.296 -> [0;32mI (118) boot_comm: chip revision: 3, min. application chip revision: 0[0m
```

Program with: PMD_WiFiReady_Elegant_migratingtocpph
Serial Monitor

ForrestErickson, 2025-04-10:
**US0004**
Serial Monitor of new never programed part

19:37:10.027 -> ets Jul 29 2019 12:21:46
19:37:10.027 -> 
19:37:10.027 -> rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
19:37:10.027 -> configsip: 0, SPIWP:0xee
19:37:10.027 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:37:10.027 -> mode:DIO, clock div:2
19:37:10.027 -> load:0x3fff0018,len:4
19:37:10.027 -> load:0x3fff001c,len:6932
19:37:10.027 -> load:0x40078000,len:13712
19:37:10.027 -> load:0x40080400,len:4260
19:37:10.027 -> entry 0x400806e4
19:37:10.027 -> [0;32mI (71) boot: Chip Revision: 3[0m
19:37:10.027 -> [0;32mI (72) boot_comm: chip revision: 3, min. bootloader chip revision: 0[0m
19:37:10.027 -> [0;32mI (40) boot: ESP-IDF v4.0.1-193-ge7ac221 2nd stage bootloader[0m
19:37:10.027 -> [0;32mI (40) boot: compile time 02:47:00[0m
19:37:10.027 -> [0;32mI (40) boot: Enabling RNG early entropy source...[0m
19:37:10.027 -> [0;32mI (46) boot: SPI Speed      : 40MHz[0m
19:37:10.027 -> [0;32mI (50) boot: SPI Mode       : DIO[0m
19:37:10.027 -> [0;32mI (54) boot: SPI Flash Size : 4MB[0m
19:37:10.027 -> [0;32mI (58) boot: Partition Table:[0m
19:37:10.027 -> [0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
19:37:10.027 -> [0;32mI (69) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
19:37:10.027 -> [0;32mI (76) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
19:37:10.027 -> [0;32mI (84) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
19:37:10.027 -> [0;32mI (91) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
19:37:10.027 -> [0;32mI (99) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
19:37:10.027 -> [0;32mI (106) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
19:37:10.027 -> [0;32mI (114) boot: End of partition table[0m
19:37:10.027 -> [0;32mI (118) boot_comm: chip revision: 3, min. application chip revision: 0[0m

Program with: PMD_WiFiReady_Elegant_migratingtocpph
Serial Monitor

ForrestErickson, 2025-04-11:
**US0005**
Serial Monitor of new never programed part

```
19:53:27.893 -> [0;32mI (40) boot: Enabling RNG early entropy source...[0m
19:53:27.893 -> [0;32mI (46) boot: SPI Speed      : 40MHz[0m
19:53:27.893 -> [0;32mI (50) boot: SPI Mode       : DIO[0m
19:53:27.893 -> [0;32mI (54) boot: SPI Flash Size : 4MB[0m
19:53:27.893 -> [0;32mI (58) boot: Partition Table:[0m
19:53:27.893 -> [0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
19:53:27.893 -> [0;32mI (69) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
19:53:27.893 -> [0;32mI (76) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
19:53:27.893 -> [0;32mI (84) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
19:53:27.893 -> [0;32mI (91) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
19:53:27.893 -> [0;32mI (99) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
19:53:27.893 -> [0;32mI (106) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
19:53:27.893 -> [0;32mI (114) boot: End of partition table[0m
19:53:27.893 -> [0;32mI (118) boot_comm: chip revision: 3, min. application chip revision: 0[0m
19:53:27.893 -> [0;32mI (125) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x29230 (168496) map[0m
19:53:27.973 -> [0;32mI (194) esp_image: segment 1: paddr=0x00129258 vaddr=0x3ffbdb60 size=0x03974 ( 14708) load[0m
19:53:27.973 -> [0;32mI (201) esp_image: segment 2: paddr=0x0012cbd4 vaddr=0x40080000 size=0x00400 (  1024) load[0m
19:53:27.973 -> [0;32mI (202) esp_image: segment 3: paddr=0x0012cfdc vaddr=0x40080400 size=0x03034 ( 12340) load[0m
19:53:27.973 -> [0;32mI (215) esp_image: segment 4: paddr=0x00130018 vaddr=0x400d0018 size=0x1068cc (1075404) map[0m
19:53:28.345 -> [0;32mI (604) esp_image: segment 5: paddr=0x002368ec vaddr=0x40083434 size=0x195c8 (103880) load[0m
19:53:28.423 -> [0;32mI (647) esp_image: segment 6: paddr=0x0024febc vaddr=0x400c0000 size=0x00064 (   100) load[0m
19:53:28.423 -> [0;32mI (665) boot: Loaded app from partition at offset 0x100000[0m
19:53:28.423 -> [0;32mI (665) boot: Disabling RNG early entropy source...[0m
19:53:28.559 -> BLUFI BLE is not connected yet
19:53:28.559 -> 2.1.0

19:53:28.559 -> max tx power=78,ret=0

19:55:37.094 -> ets Jul 29 2019 12:21:46
19:55:37.094 -> 
19:55:37.094 -> rst:0x1 (POWERON_RESET),boot:0x7 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
19:55:37.094 -> waiting for download

```

After rebooting, accessing the soft AP, Setting WiFi credientials
```
19:59:17.327 -> load:0x40078000,len:16460
19:59:17.327 -> load:0x40080400,len:4
19:59:17.327 -> load:0x40080404,len:3504
19:59:17.327 -> entry 0x400805cc
19:59:17.726 -> No saved WiFi credentials found.
19:59:17.726 -> *wm:AutoConnect 
19:59:17.802 -> *wm:No wifi saved, skipping 
19:59:17.802 -> *wm:AutoConnect: FAILED 
19:59:17.848 -> *wm:StartAP with SSID:  ESP32-Setup
19:59:18.317 -> *wm:AP IP address: 192.168.4.1
19:59:18.317 -> *wm:Starting Web Portal 
19:59:56.757 -> *wm:Connecting to NEW AP: VRX
19:59:56.757 -> *wm:connectTimeout not set, ESP waitForConnectResult... 
19:59:59.762 -> *wm:Connect to new AP [SUCCESS] 
19:59:59.762 -> *wm:Got IP Address: 
19:59:59.762 -> *wm:192.168.1.184 
20:00:00.828 -> *wm:config portal exiting 
20:00:00.828 -> Failed to save WiFi credentials.
20:00:00.828 -> Connected to WiFi!
20:00:00.828 -> ===================================
20:00:00.828 ->  V0.0.1 
20:00:00.828 -> Model: HW2_WiFiReady_Elegant
20:00:00.828 -> SN: 00001
20:00:00.828 -> Compiled at: Apr 10 2025 12:55:59
20:00:00.828 -> GNU Affero General Public License, version 3 
20:00:00.828 -> ===================================
20:00:00.828 -> 
20:00:00.828 -> Initializing DFPlayer module. UART2 Begin...
20:00:00.828 -> DFPlayer begin with ACK and Reset
20:00:04.543 -> Begin DFPlayer again.

```

nk25719, 2025-04-14:
LB0003:

```

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6932
load:0x40078000,len:13712
load:0x40080400,len:4260
entry 0x400806e4
[0;32mI (71) boot: Chip Revision: 3[0m
[0;32mI (72) boot_comm: chip revision: 3, min. bootloader chip revision: 0[0m
[0;32mI (40) boot: ESP-IDF v4.0.1-193-ge7ac221 2nd stage bootloader[0m
[0;32mI (40) boot: compile time 02:47:00[0m
[0;32mI (40) boot: Enabling RNG early entropy source...[0m
[0;32mI (46) boot: SPI Speed      : 40MHz[0m
[0;32mI (50) boot: SPI Mode       : DIO[0m
[0;32mI (54) boot: SPI Flash Size : 4MB[0m
[0;32mI (58) boot: Partition Table:[0m
[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (69) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (76) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (84) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32mI (91) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
```

ForrestErickson, 2025-04-27:
This is enough.

Closing.

CC: @nk25719

closed 2025-04-27 by ForrestErickson