PubInv/krake #176: Hardware, PCB Rev 2, Add another speaker connector for 0.1 spacing or larger.

ForrestErickson, 2025-04-13:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex.

 I'm always frustrated when I cannot connect to the tiny connector on the Krake speaker.
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

Add a 0.1 spacing header in parallel for Dupont like connectors. Perhaps a locking ramp too.

Possible consider to add speaker spade connections.

ForrestErickson, 2025-08-01:
## 1.25mm Speaker Header
See Issue #231 for notes on a part number for a connector compatible with the 28mm speaker.

nk25719, 2025-08-08:
## Added JST 2.00 mm 2pin Header Speaker Connector 

[image]

## Connected the connector of the external speaker J603 parallel to the connector of the internal speaker J601
[image]

closed 2025-08-08 by nk25719