PubInv/krake #180: Lee and Nagham Do Not Understand Scope for #define for PROG_NAME

ForrestErickson, 2025-04-17:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
 
Lee and Nagham were making a sketch for testing the COM port. 
We started with the sketch PMD_WiFiReady_Elegant_migratingtocpph.
We went to rename the PROG_NAME (and others) and found it is not unique.
We tried to make a single #define PROG_NAME in the main ino sketch with out success.

[image]
**See lines 2 and 3 of this screen shot.**

 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 
Lee and Nagham would like Rob's help developing and understanding a way to make a global PROG_NAME which all files can use.
We want to change in one and only one place the PROG_NAME.

ForrestErickson, 2025-04-22:
@RobertLRead 

This is one of the items about scope I wanted to learn.

ForrestErickson, 2025-04-22:
DEBUG is also not global and is defined in several files.

RobertLRead, 2025-04-22:
PROG_NAME is defined on line #28 in the file gpad_utility.h.

It will be in scope in any file textually after the #include of it like:

#include "gpad_utility.h"

That is why it IS in scope when used on line #205 of GPAD_API.ino.