PubInv/krake #181: USB to RS232 Enumeration, Arduino IDE and RealTerminal Not Connect

ForrestErickson, 2025-04-18:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
 
Nagham has a USB to RS232 adapter which the Arduino would not connect.

 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 
 
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
 
 
**Additional context**
_Add any other context or screenshots about the feature request here._

ForrestErickson, 2025-04-18:
Advanced Settings on Windows 10

[image]

nk25719, 2026-10-18:
Added to the [technical manual troubleshooting section](https://docs.google.com/document/d/150WA6Mb1_SFOBmFc9qBpT5b-MqXb7ejaZbKIxUqaDBY/edit?tab=t.0#heading=h.x3qw1tfpmsvg)

closed 2026-10-18 by nk25719