PubInv/krake #187: Attachment of Speaker in Krake Enclosure

ForrestErickson, 2025-04-25:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
 
The 28mm speaker needs an attachement to the shell.
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 Holde it with spring clip, screws or other temporary measures.
 
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

Lee hot glued one speaker in place

[image]

ForrestErickson, 2025-04-26:
@RobertLRead 

Regarding the potential new mechanical engineering help this is something we have yet to do.
I am not sure making a mount for the speaker is a good learning lesson but think we should show him and you what we (@nk25719 and I) are doing.

Lee

nk25719, 2025-04-26:
[image]

The mounting for the speaker was done today, we need to print and test the fit. 

[image]

nk25719, 2025-11-21:
Refer to #141

closed 2025-11-21 by nk25719