PubInv/krake #188: DOC, Procedure for Loading MP3 Files in SD Card

ForrestErickson, 2025-04-25:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
 
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 
 
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
 
 
**Additional context**
_Add any other context or screenshots about the feature request here._
Screen shot of Lee's card of 20250425
[image]

ForrestErickson, 2025-04-25:
To match the alarm levels 0-5, the file should be:

1. OK
2. Information
3. Problem
4. Waring
5. Critical
6. Panic
7. Muted
8. Unmuted
9. Help
10. ...

nk25719, 2025-04-27:
1. Format the SD Card

- Use FAT16 (if the SD card is 2GB or less)
- Use FAT32 (for SD cards larger than 2GB, like 8GB, 16GB, 32GB)

 2. Organize Your Files 

- If you  want files: Always 4 digits for best compatibility.

 ```
 0001.mp3
0002.mp3
0003.mp3
```

 -  If you want folders: 2 digits; File names = 3 digits inside folders.

```
(MicroSD Root)
├── 0001.mp3
├── 0002.mp3
├── 0003.mp3
├── 01/
│    ├── 001.mp3
│    ├── 002.mp3
│    └── 003.mp3
├── 02/
│    ├── 001.mp3
│    ├── 002.mp3
│    └── 003.mp3

```

> [!IMPORTANT]
>  - File/ Folder Names should only contain numbers. No spaces, no letters, no special characters.
> ### Maximum number of files: 
> - 3000 files in root. 
> - 255 folders max, each folder can hold up to 255 files.

ForrestErickson, 2026-01-05:
The file: [copy_to_sd.sh](https://github.com/user-attachments/files/24440300/copy_to_sd.sh)

Closing as done.

closed 2026-01-05 by ForrestErickson