PubInv/krake #194: Find Used Network IP Address

ForrestErickson, 2025-05-04:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
 
I am frustrated when I do not know the IP Address of a Krake.
It would be nice if I could use "arp -a " to narrow the search but if a device has not been accessed by the router for some time it is dropped from the arp table.

 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
A utility to quickly ping every address.

ForrestErickson, 2025-05-04:
## A Batch file to ping all
file name "pingall.bat"

```
REM pingall.bat

@echo off
Title Piniing all IP Addrsses & Mode 70,4 & color 0B
::-------------------------------------------------------------------------------------
REM First We Check The Status Of The Internet Connection
Call :Check_Connection
::-------------------------------------------------------------------------------------
:Main
Title Ping the IP Address

mode con: cols=160 lines=78

arp -a

FOR /L %%j IN (1, 1, 250) DO (
    echo Number: %%j
REM	Test-Connection -Ping -TargetName 192.168.1.%%j
REM	'Test-Connection' -Ping -TargetName 192.168.1.%%j
start ping -n 1 192.168.1.%%j

)

arp -a

pause

```

nk25719, 2025-07-05:
https://github.com/PubInv/krake/tree/main/PMD/PingAll

ForrestErickson, 2025-07-06:
Closing as done.

closed 2025-07-06 by ForrestErickson