PubInv/krake #196: Journal, First WiFi Connection

nk25719, 2025-05-09:
07:26:34.115 -> rst:0x1 (POWERON_RESET),boot:0x7 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
07:26:34.147 -> waiting for download
07:29:13.227 -> ets Jul 29 2019 12:21:46
07:29:13.227 -> 
07:29:13.227 -> rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
07:29:13.227 -> configsip: 0, SPIWP:0xee
07:29:13.227 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
07:29:13.227 -> mode:DIO, clock div:1
07:29:13.227 -> load:0x3fff0030,len:4832
07:29:13.297 -> load:0x40078000,len:16460
07:29:13.297 -> load:0x40080400,len:4
07:29:13.297 -> load:0x40080404,len:3504
07:29:13.297 -> entry 0x400805cc
07:29:13.747 -> ===================================
07:29:13.747 -> PubInv 
07:29:13.747 -> KRAKE_
07:29:13.747 -> GPAD_API 0.39 
07:29:13.747 -> Builtin ESP32 MAC Address: 
07:29:13.747 -> Alarm Topic: 
07:29:13.747 -> Broker: public.cloud.shiftr.io
07:29:13.747 -> Compiled at: May  9 2025 07:27:56
07:29:13.747 -> GNU Affero General Public License, version 3 
07:29:13.781 -> ===================================
07:29:13.781 -> 
07:29:17.256 -> 
07:29:17.256 -> PubInv GPAD: enter command in form CDa (C is a char, D is a digit)
07:29:17.256 -> MAC: 
07:29:17.256 -> 
07:29:17.256 -> Starting WiFi as STA
07:29:17.321 -> My mac is 14:2b:2f:eb:1e:24
07:29:17.321 -> MAC as char array: 142B2FEB1E24
07:29:18.497 -> No saved WiFi credentials found.
07:29:18.497 -> *wm:AutoConnect 
07:29:18.497 -> *wm:No wifi saved, skipping 
07:29:18.528 -> *wm:AutoConnect: FAILED for  1 ms
07:29:18.528 -> *wm:StartAP with SSID:  ESP32-Setup
07:29:19.017 -> *wm:AP IP address: 192.168.4.1
07:29:19.017 -> *wm:Starting Web Portal 
07:30:40.658 -> *wm:5 networks found

nk25719, 2025-05-09:
07:34:06.023 -> *wm:Connecting to NEW AP: Home
07:34:06.023 -> *wm:connectTimeout not set, ESP waitForConnectResult... 
07:34:11.129 -> *wm:Connect to new AP [SUCCESS] 
07:34:11.129 -> *wm:Got IP Address: 
07:34:11.129 -> *wm:192.168.10.156 
07:34:12.139 -> *wm:config portal exiting 
07:34:12.139 -> Failed to save WiFi credentials.
07:34:12.139 -> Connected to WiFi!
07:34:12.139 -> E (298503) esp_littlefs: ./managed_components/joltwallet__littlefs/src/littlefs/lfs.c:1369:error: Corrupted dir pair at {0x0, 0x1}
07:34:12.170 -> 
07:34:12.170 -> E (298509) esp_littlefs: mount failed,  (-84)
07:34:12.170 -> E (298513) esp_littlefs: Failed to initialize LittleFS
07:34:12.170 -> E (310983) task_wdt: esp_task_wdt_reset(763): task not found
07:34:12.170 -> E (310989) task_wdt: esp_task_wdt_reset(763): task not found
07:34:12.234 -> E (311062) task_wdt: esp_task_wdt_reset(763): task not found
07:34:12.234 -> E (311062) task_wdt: esp_task_wdt_reset(763): task not found
07:34:12.303 -> E (311129) task_wdt: esp_task_wdt_reset(763): task not found
07:34:12.303 -> E (311130) task_wdt: esp_task_wdt_reset(763): task not found
07:34:12.303 -> E (311130) task_wdt: esp_task_wdt_reset(763): task not found
07:34:12.334 -> E (311134) task_wdt: esp_task_wdt_reset(763): task not found
07:34:12.334 -> E (311140) task_wdt: esp_task_wdt_reset(763): task not found
07:34:12.334 -> Done With Setup!
07:34:12.334 -> InterruptRotator example for the RotaryEncoder library.
07:34:12.334 -> UART2 Begin
07:34:12.334 -> Begin DFPlayer: isACK true, doReset false.
07:34:12.334 -> DFPlayer Mini detected!

nk25719, 2025-05-09:
Press Reset after the WiFi Credentials are saved to the ESP32.

07:35:02.687 -> ets Jul 29 2019 12:21:46
07:35:02.687 -> 
07:35:02.687 -> rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
07:35:02.687 -> configsip: 0, SPIWP:0xee
07:35:02.687 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
07:35:02.785 -> mode:DIO, clock div:1
07:35:02.785 -> load:0x3fff0030,len:4832
07:35:02.785 -> load:0x40078000,len:16460
07:35:02.785 -> load:0x40080400,len:4
07:35:02.785 -> load:0x40080404,len:3504
07:35:02.785 -> entry 0x400805cc
07:35:03.239 -> ===================================
07:35:03.239 -> PubInv 
07:35:03.239 -> KRAKE_
07:35:03.239 -> GPAD_API 0.39 
07:35:03.239 -> Builtin ESP32 MAC Address: 
07:35:03.239 -> Alarm Topic: 
07:35:03.239 -> Broker: public.cloud.shiftr.io
07:35:03.239 -> Compiled at: May  9 2025 07:27:56
07:35:03.239 -> GNU Affero General Public License, version 3 
07:35:03.239 -> ===================================
07:35:03.239 -> 
07:35:06.730 -> 
07:35:06.730 -> PubInv GPAD: enter command in form CDa (C is a char, D is a digit)
07:35:06.730 -> MAC: 
07:35:06.730 -> 
07:35:06.730 -> Starting WiFi as STA
07:35:06.827 -> My mac is 14:2b:2f:eb:1e:24
07:35:06.827 -> MAC as char array: 142B2FEB1E24
07:35:08.001 -> No saved WiFi credentials found.
07:35:08.001 -> *wm:AutoConnect 
07:35:08.001 -> *wm:Connecting to SAVED AP: Home
07:35:08.495 -> *wm:connectTimeout not set, ESP waitForConnectResult... 
07:35:08.820 -> *wm:AutoConnect: SUCCESS 
07:35:08.820 -> *wm:STA IP Address: 192.168.10.156
07:35:08.820 -> Failed to save WiFi credentials.
07:35:08.820 -> Connected to WiFi!
07:35:08.820 -> Done With Setup!
07:35:08.820 -> InterruptRotator example for the RotaryEncoder library.
07:35:08.820 -> UART2 Begin
07:35:08.820 -> Begin DFPlayer: isACK true, doReset false.
07:35:08.820 -> DFPlayer Mini detected!

nk25719, 2025-05-09:
Presvious messages Were captured on LB0004

ForrestErickson, 2025-05-22:
Closing for now.

closed 2025-05-22 by ForrestErickson