PubInv/krake #197: Firmware, Rev 0.39, Not Responding to MQTT,

ForrestErickson, 2025-05-11:
**Describe the bug**
_A clear and concise description of what the bug is._

### Short Summary of the expected behavior:  
Respond to messages sent from PMD_Processing_MQTT

### Short Summary of the buggy behavior:
At 20250511 5:45 AM ET, Krake units SNs: KRAKE US0001, KRAKE US0002, and KRAKE US0005 were powered but not responding to any PMD_Processing MQTT commands.
The LCD message did not change and there was no audio message.

There was no heart beat reported on the PMD_Processing_MQTT.
[image]

Connecting to  KRAKE on COM port through a USB to COM adapter on COM7.
Got the following at the "i" information command.
1. US0005 
[image]
Audio was "OK"
2. US0002 
[image]
Audio was "Information"
3. US0001
[image]
Audio was "Information"

ForrestErickson, 2025-05-11:
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

ForrestErickson, 2025-05-11:
## Soft AP Connections
One device, SN unknown is presenting the WiFi manager
However it does not find the VRX access point.
[image]

### VRX Has Gone Offline
Lee notes that the IP address for the VRX WiFi access point is 192.168.1.253 and that is not on the list of of the "apt -a"
After resetting the VRX WiFi access point  it can be pinged.

[image]

Resetting the VRX has not fixed the problem
However still no US Krakes reporting on the MQTT status.
[image]

At 20250511 7:19 there are no ESP32 WiFi mangers found with Lee's cell phone.

ForrestErickson, 2025-05-11:
## Press Reset on US0005
To attempt to get control.
The DUT attempts to connect to readfamilynetwork It eventualy times out and presents the WiFi manager.

```
07:43:41.777 -> *wm:StartAP with SSID:  ESP32-Setup
07:43:42.278 -> *wm:AP IP address: 192.168.4.1
07:43:42.278 -> *wm:Starting Web Portal 
07:43:45.222 -> ets Jul 29 2019 12:21:46
07:43:45.223 -> 
07:43:45.223 -> rst:0x3 (SW_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
07:43:45.223 -> configsip: 0, SPIWP:0xee
07:43:45.223 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
07:43:45.223 -> mode:DIO, clock div:1
07:43:45.223 -> load:0x3fff0030,len:4604
07:43:45.223 -> ho 0 tail 12 room 4
07:43:45.223 -> load:0x40078000,len:15488
07:43:45.223 -> load:0x40080400,len:4
07:43:45.223 -> load:0x40080404,len:3180
07:43:45.223 -> entry 0x400805b8
07:43:45.768 -> ===================================
07:43:45.768 -> PubInv 
07:43:45.768 -> KRAKE_
07:43:45.768 -> GPAD_API 0.39 
07:43:45.814 -> Builtin ESP32 MAC Address: 
07:43:45.814 -> Alarm Topic: 
07:43:45.814 -> Broker: public.cloud.shiftr.io
07:43:45.814 -> Compiled at: May  7 2025 13:24:06
07:43:45.814 -> GNU Affero General Public License, version 3 
07:43:45.814 -> ===================================
07:43:45.814 -> 
07:43:49.292 -> 
07:43:49.292 -> PubInv GPAD: enter command in form CDa (C is a char, D is a digit)
07:43:49.292 -> MAC: 
07:43:49.292 -> 
07:43:49.292 -> Starting WiFi as STA
07:43:49.415 -> My mac is ec:c9:ff:7d:8e:e8
07:43:49.415 -> MAC as char array: ECC9FF7D8EE8
07:43:50.634 -> No saved WiFi credentials found.
07:43:50.634 -> *wm:AutoConnect 
07:43:50.634 -> *wm:Connecting to SAVED AP: readfamilynetwork
07:43:51.104 -> *wm:connectTimeout not set, ESP waitForConnectResult... 
07:43:53.605 -> *wm:AutoConnect: FAILED for  3005 ms
07:43:53.605 -> *wm:StartAP with SSID:  ESP32-Setup
07:43:54.100 -> *wm:AP IP address: 192.168.4.1
07:43:54.100 -> *wm:Starting Web Portal 
07:43:54.100 -> *wm:Starting Web Portal 
07:47:21.768 -> *wm:Connecting to NEW AP: VRX
07:47:21.840 -> *wm:connectTimeout not set, ESP waitForConnectResult... 
07:47:31.000 -> *wm:Connect to new AP [SUCCESS] 
07:47:31.061 -> *wm:Got IP Address: 
07:47:31.061 -> *wm:192.168.1.184 
07:47:32.005 -> *wm:config portal exiting 
07:47:32.005 -> Failed to save WiFi credentials.
07:47:32.037 -> Connected to WiFi!
07:47:32.037 -> Done With Setup!
07:47:32.037 -> InterruptRotator example for the RotaryEncoder library.
07:47:32.037 -> UART2 Begin
07:47:32.037 -> Begin DFPlayer: isACK true, doReset false.
07:47:32.037 -> DFPlayer Mini detected!
07:47:36.122 -> =================dfPlayer State: -1
07:47:36.185 -> dfPlayer Volume: 513
07:47:36.191 -> dfPlayer EQ: 20
07:47:36.265 -> SD Card FileCounts: 0
07:47:36.265 -> Current File Number: 10
07:47:36.347 -> File Counts In Folder: 1
07:47:36.394 -> =================Device connected at IPaddress: 192.168.1.184
07:47:36.394 -> Attempting MQTT connection...success!
07:47:37.427 -> connected!
07:47:56.382 -> Device connected at IPaddress: 192.168.1.184
07:48:16.356 -> Device connected at IPaddress: 192.168.1.184
07:48:36.385 -> Device connected at IPaddress: 192.168.1.184
07:48:56.391 -> Device connected at IPaddress: 192.168.1.184
07:49:16.347 -> Device connected at IPaddress

```
The DUT, US0005 has begun responding.
Here is what Lee things happened. 
He pressed reset.
The DUT tried and failed to connect to the Read family AP
It timed out and served the WiFi manager on 192.168.4.1.
Lee refreshed the web page on the phone connected to that web address which sent the credentials and the DUT connected to the VRX AP.

ForrestErickson, 2025-05-11:
## Possible Steps To Reproduce

1. Start with DUt working normally on a network AP
2. Interupt the network AP
3. On the USB to UART , UART0 , Check to see that the Krake has lost the WiFi and MQTT connection and is trying to connect to Readfamily network.
4. Look for a WiFi Manager soft AP.

ForrestErickson, 2025-05-11:
## Journal of Attempt to Reproduce.
AT 20250511 1121 EDT the KRAKE US0005 is on line and responding to the PMD_Processing sketch

[image]

At 11:22 disconnect the power to the VRX AP.
Right away the USB UART0 debug serial traffice indicates trying to connect to the Read family WiFi access point.
[image]

At 11:28 reconnect the power to the VRX AP.

The DUT is still trying the readfamilynetwork and not connecting to VRX.

```
Connecting to WiFi: readfamilynetwork
E (13547941) wifi:sta is connecting, return error
Failed to connect WiFi.
Attempting MQTT connection...failed, rc=-2Attempting MQTT connection...failed, rc=-2Attempting MQTT connection...failed, rc=-2failed to reconnect!

```

Checking the available WiFi networks, Lee did not find an ESP32 soft AP.

Lee concludes he has reproduced the failure. 

## GPAD_API, Rev 0.39 does not recover from a failure to connect to a WiFi. 

### Reseting DUT

```
GPAD_API 0.39
Builtin ESP32 MAC Address:
Alarm Topic:
Broker: public.cloud.shiftr.io
Compiled at: May  7 2025 13:24:06
GNU Affero General Public License, version 3
===================================

PubInv GPAD: enter command in form CDa (C is a char, D is a digit)
MAC:

Starting WiFi as STA
My mac is ec:c9:ff:7d:8e:e8
MAC as char array: ECC9FF7D8EE8
No saved WiFi credentials found.
*wm:AutoConnect
*wm:Connecting to SAVED AP: readfamilynetwork
*wm:connectTimeout not set, ESP waitForConnectResult...
*wm:AutoConnect: FAILED for  3005 ms
*wm:StartAP with SSID:  ESP32-Setup
*wm:AP IP address: 192.168.4.1
*wm:Starting Web Portal
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4604
ho 0 tail 12 room 4
load:0x40078000,len:15488
load:0x40080400,len:4
load:0x40080404,len:3180
entry 0x400805b8
===================================
PubInv
KRAKE_
GPAD_API 0.39
Builtin ESP32 MAC Address:
Alarm Topic:
Broker: public.cloud.shiftr.io
Compiled at: May  7 2025 13:24:06
GNU Affero General Public License, version 3
===================================

PubInv GPAD: enter command in form CDa (C is a char, D is a digit)
MAC:

Starting WiFi as STA
My mac is ec:c9:ff:7d:8e:e8
MAC as char array: ECC9FF7D8EE8
No saved WiFi credentials found.
*wm:AutoConnect
*wm:Connecting to SAVED AP: readfamilynetwork
*wm:connectTimeout not set, ESP waitForConnectResult...
*wm:AutoConnect: FAILED for  3004 ms
*wm:StartAP with SSID:  ESP32-Setup
*wm:AP IP address: 192.168.4.1
*wm:Starting Web Portal

```

The DUT now is offering a soft AP.

ForrestErickson, 2025-05-11:
## LED_BUILTIN Normal Behavior
Note when the DUT is operating normally the Wink LED has a regular period and duty cycle.

[image]

nk25719, 2026-01-31:
Resolved, version 0.42 no MQTT connection issue.