PubInv/krake #199: Software,

nk25719, 2025-05-12:
**Describe the bug**
_A clear and concise description of what the bug is._
Short Summary of the expected behavior:
Short Summary of the buggy behavior:
 
Processing serial monitor reporting the MAC address of a KRAKE that is not connected to power. 
KRAKE_LB0005 is off and the MAC address is showing "F024F9F1B880" as online and the RSSI is displaying. 

The KRAKE devices that are on and operating are : LB0002 and LB001

[image]

 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
 
 
**Expected behavior**
_A clear and concise description of what you expected to happen._
 
 
**Screenshots**
_If applicable, add screenshots to help explain your problem._
 
 
**Hardware description**
Model:
Serial Number:
Other:
 
**Test Equipment**
Model:
Serial Number:

**Additional context**
_Add any other context about the problem here._

nk25719, 2025-05-12:
SD card was not inserted to KRAKE LB0002, 
after inserting the SD card, the Processing is not reorting LB0002 online, but still, also, reporting LB0005 online.

[image]

ForrestErickson, 2025-05-12:
Nagham, 
Please double check the MAC address in each Krake. 
Compare against the dictionary (table) in 
Look at:  mac_to_NameDict near line 57.

[image]