PubInv/krake #201: No LCD Backlight, KRAKE US0003

ForrestErickson, 2025-05-14:
**Describe the Failure / Bug**
_A clear and concise description of what the failure / bug is. Describe the problem NOT the fix._
Short Summary of the expected behavior: The DUT should have a back light on the LCD at power on boot.

Short Summary of the buggy behavior:  No LCD back light ever.

**Device Under Test, aka DUT**
Steps to reproduce the behavior:
**Apply power to the DUT.**

 
**Hardware description**
Model:  KRAKE
Serial Number: US0003

ForrestErickson, 2025-05-14:
@RobertLRead 
Please photograph the solder connections on the pins on the LCD connection both at the KRAKE PCB and on the LCD. 

The Anode and Cathode connections from the KRAKE to the LCD are about the only places where this could go wrong.

Here is the schematic 

[image]

And here is the KRAKE PCB

[image]

CC: @nk25719

RobertLRead, 2025-05-20:
After soldering, it looks like this:

[image]

RobertLRead, 2025-05-20:
This is a "before" image of the Krake SN03 as received (the (un) solder joints):
[image]

RobertLRead, 2025-05-20:
Upon soldering, the characters of the splash screen appear and the backlight is on.

However, a new symptom has appeared, which is that SN03 is now constantly rebooting a few seconds after startup.

RobertLRead, 2025-05-20:
Note: Before I noticed that LCD was not working, we worked successfully (without reboots) on versions like 0.39 of the GPAD software (replacing the factory installed software.)