PubInv/krake #202: Develop User Interface with LCD (20x4) and Rotary Encoder

ForrestErickson, 2025-05-18:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. 
As of GPAD_API V0.39 there is no direct Krake user interface using the LCD and the Rotary Encoder.

 
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 
There are several issues identifying some use cases and some more we can identify.

- [ ] Issue #51 
- [ ] Issue #53 
- [ ] Issue #86 
- [ ] Set an override for number of times audio alarm plays. (Assumes that the PMD can set a parameter for number of times alarm plays. The local use of the KARAKE can set an override.  Set 0 for never stops repeating, 1 for repeat once and so on. Set a negative number to not play the alarm at all.)
- [ ] Show KRAKE Settings (Volume level, MAC, IP ADDRESS, WiFi Credentials, Broker URL, Subscribe and publish topics.
- [ ] Set the COM port serial port parameters, BAUD with 9600 default, Data format with 8N1 default, Flowcontrol with NONE for default,  other.....
- [ ] Set a Debug level with default 0, to make more verbose debug messages.

ForrestErickson, 2025-05-18:
## Case Study GUI
For some good ideas, here is a tutorial on a user interface menu method.
https://www.instructables.com/Arduino-Settings-Menu/

nk25719, 2025-05-27:
# 🧭 Krake Firmware Sprint — Full Feature Tracker

| Key       | Summary                                                                 | Type         | Status        | Plan |
|-----------|-------------------------------------------------------------------------|--------------|---------------|------|
| KRK-051   | Add LCD Menu to Set Alarm Mute Timeout                                  | Feature      | 🟡 In Progress | Add rotary-encoder-driven menu with EEPROM-backed timeout persistence |
| KRK-053   | Add LCD Menu to Set Speaker Volume                                      | Feature      | 🔲 To Do       | Use `dfPlayer.volume()` via encoder position |
| KRK-091   | Add Menu to Set Alarm Repeat Count (0=infinite, -1=never)               | Feature      | 🔲 To Do       | Extend alarm play logic in `playAlarmLevel()` |
| KRK-086   | Display WiFi Offline Warning                                            | Bug          | 🔲 To Do       | Use `WiFi.status()` in loop and `lcd.print()` fallback |
| KRK-066   | Human-readable MQTT Failures on Serial                                  | Enhancement  | 🔲 To Do       | Improve `interpretBuffer()` error outputs |
| KRK-093   | LED Patterns for WiFi/MQTT Connect (double/triple wink)                 | Enhancement  | 🔲 To Do       | Reuse `wink()` logic from `Wink.cpp` with event flags |
| KRK-067   | Add MQTT Last Will & Testament Support                                  | Feature      | 🔲 To Do       | Update `PubSubClient` setup with LWAT topic/status |
| KRK-101   | Show Settings (IP, MAC, Topics, Volume, WiFi, MQTT) on LCD              | Feature      | 🔲 To Do       | Create menu page with info from `WiFiManagerOTA`, `PubSubClient`, `dfPlayer` |
| KRK-154   | IP Not Shown on LCD                                                     | Bug          | ✅ Done        | Addressed in `GPAD_HAL.cpp` via `showStatusLCD()` |
| KRK-166   | Fix GPAD_API Swallowing "." and ","                                     | Bug          | 🔲 To Do       | Update `filter_control_chars()` in `GPAD_HAL.cpp` |
| KRK-110   | LCD Menu to Set Serial COM Parameters (baud, 8N1, flow)                 | Feature      | 🔲 To Do       | Add submenu under “Advanced” using encoder input |
| KRK-115   | Runtime Debug Level Setting via LCD Menu                                | Feature      | 🔲 To Do       | Add `int debugLevel = 0;` with LCD rotary control |
| KRK-130   | Add Setup UI for MQTT Broker URL, User, PW                              | Feature      | 🔲 To Do       | New menu to input/save via `LittleFS` |
| KRK-135   | Add Internal MQTT Broker Option                                         | Feature      | 🟠 Maybe Later | Use `AsyncMQTT` or `AsyncTCP` optionally |
| KRK-207   | Future-Proofing API: Identify Firmware Protocol                         | Tech Task    | 🔲 To Do       | Define a `j` and `i` protocol response format |
| KRK-208   | Add SPI Peripheral Support                                              | Feature      | 🔲 To Do       | Extend via SPI init hooks in HAL |