PubInv/krake #203: Krake reboots soon after booting

RobertLRead, 2025-05-20:
[image]

**Describe the Failure / Bug**
_A clear and concise description of what the failure / bug is. Describe the problem NOT the fix._
Short Summary of the expected behavior:
Short Summary of the buggy behavior:
 
 
**Device Under Test, aka DUT**
Steps to reproduce the behavior:
1. Model Number:
2. Serial Number(s):
3. Location where failure / bug occurred
4. Hardware configuration. (How connected, any accessories, what else it takes to replicate the system in which the failure occurred.)
5. Extent of Failure
** Is failure intermittent?
** Is failure fixed by power cycling?
** Is failure permanent?
 
 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
 
 
**Detailed Expected behavior**
_A clear and concise description of what you expected to happen._
 
 
**Detailed Actual, "Buggy" behavior**
_A clear and concise description of the buggy behavior._
 
 
**Screenshots**
_If applicable, add screenshots to help explain your problem._
 
 
**Hardware description**
Model:
Serial Number:
Other:
 
 
**Test Equipment**
Model:
Serial Number:
 
 
**Additional context**
_Add any other context about the problem here._

[image]

ForrestErickson, 2025-05-20:
## Normal Boot up Communication on Serial Monitor 
DUT is KRAKE SN: US0005, powered from a wall transformer and with USB to UART Adapter.

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4832
load:0x40078000,len:16460
load:0x40080400,len:4
load:0x40080404,len:3504
entry 0x400805cc
===================================
PubInv 
KRAKE_
GPAD_API 0.39 
Builtin ESP32 MAC Address: 
Alarm Topic: 
Broker: public.cloud.shiftr.io
Compiled at: May 12 2025 15:10:05
GNU Affero General Public License, version 3 
===================================

PubInv GPAD: enter command in form CDa (C is a char, D is a digit)
MAC: 

Starting WiFi as STA
My mac is ec:c9:ff:7d:8e:e8
MAC as char array: ECC9FF7D8EE8
No saved WiFi credentials found.
*wm:AutoConnect 
*wm:Connecting to SAVED AP: VRX
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address: 192.168.1.184
Failed to save WiFi credentials.
Connected to WiFi!
Done With Setup!
InterruptRotator example for the RotaryEncoder library.
```

RobertLRead, 2025-05-20:
Note: After commenting out ALL use of the LCD init and backlight, the buggy behavior goes away---that is, the system does NOT erroneously reboot.

RobertLRead, 2025-05-20:
note: The buggy behavior is removed by NOT turnning on the splash (BUT the LCD does work when invoked later.)  This change:

```
diff --git a/Firmware/GPAD_API/GPAD_HAL.cpp b/Firmware/GPAD_API/GPAD_HAL.cpp
index 122d14f..d23b7aa 100644
--- a/Firmware/GPAD_API/GPAD_HAL.cpp
+++ b/Firmware/GPAD_API/GPAD_HAL.cpp
@@ -327,7 +327,7 @@ void GPAD_HAL_setup(Stream *serialport) {
 #if (DEBUG > 0)
   serialport->println(F("Start LCD splash"));
 #endif
-  splashLCD();
+//  splashLCD();
 #if (DEBUG > 0)
   serialport->println(F("EndLCD splash"));
 #endif
@@ -541,6 +541,7 @@ void clearLCD(void) {
 
 //Splash a message so we can tell the LCD is working
 void splashLCD(void) {
+  return;
   lcd.init();  // initialize the lcd
   // Print a message to the LCD.
   lcd.backlight();
@@ -662,10 +663,8 @@ void unchanged_anunicateAlarmLevel(Stream *serialport) {
 
     //   serialport->print("note lvl");
     //   serialport->println(note_lvl);
-
     tone(TONE_PIN, BUZZER_LVL_FREQ_HZ[note_lvl], INF_DURATION);
 
-
   } else {
     noTone(TONE_PIN);
   }
```
accomplishes this, which is simply commoing out the "splashLCD" call at line 330 in the GPAD_HAL.cpp file.

ForrestErickson, 2025-05-21:
## Here is how to measure the 3.3V supply

Instructions.

The pink trace is the +3.3. Sorry we forgot to put a test point on it.
You can probe it at the pad of D103 for example or on the second hole down to the left of the ESP32.  
Ground test point is across the board on the right.  
[image]

I held the probe on the PCB with my left hand and then applied power to the device with my right hand and then when the trace filled the screen I pressed the Run/Stop button on the oscilloscope.

RobertLRead, 2025-05-22:
I was not able to get my oscilloscope to trigger properly. I took the following image by pushing the "force" button on my oscilloscope very quickly after hitting the "reset" button near the ESP32.  It looks pretty similar to yours. 

[image]

@ForrestErickson I am unsure of the next step.

ForrestErickson, 2025-07-07:
Lee googled, "what can cause "SW_RESET" on ESP32"
An AI came back with answers amount which is:

> Code Errors:
> Bugs or errors in the user's code, such as calling functions before they are initialized (e.g., ledcWrite before ledcSetup), can cause panics and resets. Specifically, functions like xQueueGenericReceive() often panic if they receive a corrupted or uninitialized queue. 

This lead me to this thread involving the setup of an LCD.
https://esp32.com/viewtopic.php?t=20073#:~:text=Re:%20SW_RESET...,to%20figure%20out%20Please%20help&text=You%20read%20these%20things%20from,task%20*after*%20starting%20Task1code?)

with the following:

>  Seems you started a task (as the stack trace starts with vPortTaskWrapper) called Task1code. This task then called into ledcWrite(), which internally called xQueueGenericReceive(). The xQueueGenericReceive() did not like the parameters it was passed or something so it called abort(), leading to the panic.
> 
> Functions like xQueueGenericReceive mostly panic if the queue they're given is corrupted or not initialized. Possibly ledcWrite was invoked before the ledc was initialized? (For instance, do you initialize it in your main task *after* starting Task1code?)
> 

HOWEVER I infer that this thread us about a case where a Real Time Operating System was involved and the root cause was the code to execute in an unexpected sequence.

ForrestErickson, 2025-07-14:
## LCD Back Light Color

> On Sat, Jul 12, 2025 at 6:51 PM FORREST ERICKSON <[forresterickson@comcast.net](mailto:forresterickson@comcast.net)> wrote:
> Robert,
>  
> So far I have failed to reproduce the SW_RESET boot loop.
> I have tested with 0.44 and went back with 0.39 on Krake LB0004 with no success.
>  
> Can you tell us the color of the LCD back light on the Krake US0003?
> We built some with Yellow/Green and some with Blue.
> Please advise when you can.
>  
> Thanks,
> Lee
> 

On 07/14/2025 11:49 AM EDT Robert Read <read.robert@gmail.com> wrote:

> It is blue.

ForrestErickson, 2025-07-14:
@RobertLRead 

Please get the code to where you can reproduce the SW_RESET bug.

Then try adding code to initialize the LCD display 
Then a delay of TBD
BEFORE any calls to clear the LCD or the splashLCD.

See the image below.
The LCD should be initialized with delay before line 386.

[image]

CC: @nk25719