PubInv/krake #205: MQTT Received Message Delay and Timing Jitter different in Lebanon and Maryville TN

ForrestErickson, 2025-05-22:
**Describe the bug**
_A clear and concise description of what the bug is._
Short Summary of the expected behavior:  Publish MQTT message get to destination with low delay
Short Summary of the buggy behavior:  At Lee's internet provider there is a typical 400 to 500 mS turn around time, however at Nagam's internet provider there is a 1000 to 5000 mS turn around time.

 
 
**To Reproduce**
Steps to reproduce the behavior:
1. Load the firmware: MQTT_timing into an ESP32.
2. Set the credentials for the WiFi LAN
3. Confirm network connection by watching the Serial Monitor
4. Capture with Serial Ploter the third parameter.
 

 
**Desired behavior**
_A clear and concise description of what you expected to happen._
Low latency
Low timing jitter 
 
**Screenshots**
_If applicable, add screenshots to help explain your problem._
 
Nagham's results

[image]

Lee's results

[image]

RobertLRead, 2025-05-26:
Here is my data --- it seems to be pretty fast. @nk25719 @ForrestErickson 

[image]

I don't know what the next step here is.

ForrestErickson, 2025-05-26:
Regarding, "I don't know what the next step here is."
I have referenced this issue in the larger issue of #202 where there is the feature request of a local Broker.

A question is would the 5-6 second latency and uncertainty at Nagham's location make the product unusable for some, many or all use cases?