PubInv/krake #208: Firmware, GPAD_API, Add Support for SPI Peripherial Interface.

ForrestErickson, 2025-05-26:
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

The Krake needs a SPI Peripheral interface. 
 
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
 
 
**Additional context**
_Add any other context or screenshots about the feature request here._

nk25719, 2025-05-27:
# 🧭 Krake Firmware Sprint — Full Feature Tracker

| Key       | Summary                                                                 | Type         | Status        | Plan |
|-----------|-------------------------------------------------------------------------|--------------|---------------|------|
| KRK-051   | Add LCD Menu to Set Alarm Mute Timeout                                  | Feature      | 🟡 In Progress | Add rotary-encoder-driven menu with EEPROM-backed timeout persistence |
| KRK-053   | Add LCD Menu to Set Speaker Volume                                      | Feature      | 🔲 To Do       | Use `dfPlayer.volume()` via encoder position |
| KRK-091   | Add Menu to Set Alarm Repeat Count (0=infinite, -1=never)               | Feature      | 🔲 To Do       | Extend alarm play logic in `playAlarmLevel()` |
| KRK-086   | Display WiFi Offline Warning                                            | Bug          | 🔲 To Do       | Use `WiFi.status()` in loop and `lcd.print()` fallback |
| KRK-066   | Human-readable MQTT Failures on Serial                                  | Enhancement  | 🔲 To Do       | Improve `interpretBuffer()` error outputs |
| KRK-093   | LED Patterns for WiFi/MQTT Connect (double/triple wink)                 | Enhancement  | 🔲 To Do       | Reuse `wink()` logic from `Wink.cpp` with event flags |
| KRK-067   | Add MQTT Last Will & Testament Support                                  | Feature      | 🔲 To Do       | Update `PubSubClient` setup with LWAT topic/status |
| KRK-101   | Show Settings (IP, MAC, Topics, Volume, WiFi, MQTT) on LCD              | Feature      | 🔲 To Do       | Create menu page with info from `WiFiManagerOTA`, `PubSubClient`, `dfPlayer` |
| KRK-154   | IP Not Shown on LCD                                                     | Bug          | ✅ Done        | Addressed in `GPAD_HAL.cpp` via `showStatusLCD()` |
| KRK-166   | Fix GPAD_API Swallowing "." and ","                                     | Bug          | 🔲 To Do       | Update `filter_control_chars()` in `GPAD_HAL.cpp` |
| KRK-110   | LCD Menu to Set Serial COM Parameters (baud, 8N1, flow)                 | Feature      | 🔲 To Do       | Add submenu under “Advanced” using encoder input |
| KRK-115   | Runtime Debug Level Setting via LCD Menu                                | Feature      | 🔲 To Do       | Add `int debugLevel = 0;` with LCD rotary control |
| KRK-130   | Add Setup UI for MQTT Broker URL, User, PW                              | Feature      | 🔲 To Do       | New menu to input/save via `LittleFS` |
| KRK-135   | Add Internal MQTT Broker Option                                         | Feature      | 🟠 Maybe Later | Use `AsyncMQTT` or `AsyncTCP` optionally |
| KRK-207   | Future-Proofing API: Identify Firmware Protocol                         | Tech Task    | 🔲 To Do       | Define a `j` and `i` protocol response format |
| KRK-208   | Add SPI Peripheral Support                                              | Feature      | 🔲 To Do       | Extend via SPI init hooks in HAL |

nk25719, 2025-06-26:
Following example: 

https://randomnerdtutorials.com/esp32-spi-communication-arduino/

ForrestErickson, 2025-06-26:
Another now closed issue, #47 referred me to this library:
https://github.com/hideakitai/ESP32DMASPI

See the ESP32DMA library which can be installed through the library manager:
[image]

ForrestErickson, 2025-06-27:
## Notes on 20250627
Developed two sketches based on ESP32SPISlave library examples.
- transfer_one_by_one_master_20250627
- transfer_one_by_one_slave_20250627

## Loop Back Testing MASTER,
Load code transfer_one_by_one_master_20250627 into an ESPDev Kit. 
Run code. 
### Loop Back Results.
Initially the code returns error.
Add a jumper from GPIO19 to GPIO 23.   
[image]

Code returns "successfully received expected data from slave"
[image]

ForrestErickson, 2025-07-01:
@RobertLRead 
This is the issue which should remind you for adding the ESP32 SPI Peripheral feature to the GPAD_API