PubInv/krake #230: Firmware, Add To MENU a RESET selection.

ForrestErickson, 2025-06-29:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. 
I'm always frustrated when I have to pull a power cord to reset the device.
 
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Add to the MENU system  a selection which will reset the ESP32 controller.
 
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
 
We could also extend the API for a reset command too.

nk25719, 2026-03-02:
Should not be part of the menu, 
but maybe an option that one can access after entering certain credentials.

ForrestErickson, 2026-03-04:
Regarding, 

> Should not be part of the menu, but maybe an option that one can access after entering certain credentials.

Please explain reasoning.

Having to pull power to reset a device is a well known bad product design. 
Reset must also be such that it is not accidently used.

## From an AI on pulling power
[image]

nk25719, 2026-03-05:
Easy access to the reset as another option of shelving or dismiss or complete is not a great way, one can easily select it by mistake; 
Usually:
Such options after selecting it, should prompt you with a message like "are you sure" or "please enter password".