PubInv/krake #242: Hardware, Enclosure, No Access To Reset, No Access to Boot Button, Enclosure Rev ???

ForrestErickson, 2025-07-14:
**Describe the Failure / Bug**
_A clear and concise description of what the failure / bug is. Describe the problem NOT the fix._

Short Summary of the expected behavior:  User may need to reset the DUT. User may need to put the DUT into Boot mode by pressing and holding Boot button and Pressing and releasing the Reset button. 

Short Summary of the buggy behavior:
The user must remove the very poorly named "Top" cover of the enclosure to do the above.

ForrestErickson, 2025-11-13:
## A Problem, Volunteers in UK and EG do not (as of 2025 Nov) have printed wiring assemblies PWAs
With out PWAs it is hard to visualize the difficult access to the Reset button and the Boot Button.

A possible solution is for the volunteer to 3D print up a model of the PWA.
It could be sliced into two parts through the PCB and glued into one body.
There is a STEP file at:  \krake\enclosure\KRAKE_PWArev1.step

Screen shot of the assembly STEP file in a slicer

[image]

Cut in to two parts

[image]

Select and flip as necessary and use auto place
[image]

Glue these two bodies to make a 3D of the PWA.

Here is a sliced file:

[Sliced_KRAKE_PWArev1.zip](https://github.com/user-attachments/files/23532504/Sliced_KRAKE_PWArev1.zip)

ForrestErickson, 2025-12-03:
The enclosure should have text to identify the buttons.

[image]

There is a "RESET" button and a "BOOT" button. See in the above screen shot.

Yhya99, 2026-01-09:
### changing the cover shape and included it to the top shell 

[image]

[image]

### cross section view of the cover 

[image]

nk25719, 2026-01-31:
Solved in the pull request #327 
https://github.com/PubInv/krake/pull/327

[image]

[image]

closed 2026-01-31 by nk25719